ISPC-TST-EM-2023/proyecto-1-grupo-06 #48: Uh#1Tk#2Malvina : Actualizar documentación ESP32

CACHITO-13, 2023-04-27:
## Actualizar en carpeta de Bibliografía los recursos del ESP32

PROFMALVINA, 2023-04-27:
[ESP32_MANUAL.pdf](https://github.com/ISPC-TST-ELECTRONICA-MICROCONTROLADA/proyecto-1-grupo-06/files/11338360/ESP32_MANUAL.pdf)

[ESP 32.pdf](https://github.com/ISPC-TST-ELECTRONICA-MICROCONTROLADA/proyecto-1-grupo-06/files/11338362/ESP.32.pdf)

[PIN OUT ESP 32.docx](https://github.com/ISPC-TST-ELECTRONICA-MICROCONTROLADA/proyecto-1-grupo-06/files/11338363/PIN.OUT.ESP.32.docx)

PROFMALVINA, 2023-04-27:
Javier , completé la tarea Actualizar documentación ESP32.

CACHITO-13, 2023-04-27:
Hola Malvina, ¿cómo estás?
Te comento que los documentos no están relacionados con el proyecto (UPS - TTS056.pdf). Dejo una captura del mismo.
Además están alojados en main, debés moverlos a las carpetas correspondientes.
Saludos, Javier

CACHITO-13, 2023-04-27:
[image]

PROFMALVINA, 2023-05-01:
Javier tarea corregida.

CACHITO-13, 2023-05-01:
Buenísimo, gracias Malvina! Buen trabajo.

closed 2023-05-01 by CACHITO-13