ISPC-TST-EM-2023/proyecto-1-grupo-06 #49: Uh#1Tk#3Malvina : Actualizar Bibliotecas / Recursos

CACHITO-13, 2023-04-27:


PROFMALVINA, 2023-04-27:
[UPS - TTS056.pdf](https://github.com/ISPC-TST-ELECTRONICA-MICROCONTROLADA/proyecto-1-grupo-06/files/11338382/UPS.-.TTS056.pdf)

CACHITO-13, 2023-04-27:
Hola Malvina, ¿cómo estás?
Te comento que los documentos no están relacionados con el proyecto (UPS - TTS056.pdf). Dejo una captura del mismo.
Además están alojados en main, debés moverlos a las carpetas correspondientes.
Los documentos no deberías subirlos en los Issues, sino al repositorio.
Saludos, Javier

CACHITO-13, 2023-04-27:
[image]

PROFMALVINA, 2023-05-01:
Javier, corregí lo que me solicitaste.

CACHITO-13, 2023-05-01:
Buenísimo, gracias Malvina! Buen trabajo.

closed 2023-05-01 by CACHITO-13