ISPC-TST-EM-2023/proyecto-1-grupo-06 #51: Uh#1Tk#2Malvina: Mover los documentos subidos a main a las carpetas correspondientes

CACHITO-13, 2023-04-27:
# Tarea designada por Scrum master

1. Comprobar la utilidad de los docuentos subidos al repositorio y filtrar los que no aportan al trabajo cooperativo (**NO SUBIR DOCUMENTOS AJENOS AL PROYECTO**)
2. Mover los documentos ya filtrados de **main** a las carpetas correspondientes (B_Bibliografía o C_Ampliación)
3. Asegurarse de dar el uso correcto a las **Issues** del proyecto, teniendo en claro el objetivo de las mismas en el proyecto.

PROFMALVINA, 2023-05-01:
Javier, documentos movidos.

CACHITO-13, 2023-05-01:
Malvina por favor asignate esta Issue acá o en el tablero Kanban y quedaría lista. 
¡Muchas gracias!

PROFMALVINA, 2023-05-01:
Javier, tarea corregida.

CACHITO-13, 2023-05-01:
Buenísimo Malvina. Cierro la Issue como completada. ¡Muchas gracias!

[image]

closed 2023-05-01 by CACHITO-13

PROFMALVINA, 2023-05-01:
> <ISPC-TST-ELECTRONICA-MICROCONTROLADA/proyecto-1-grupo-06/issues/51/1529304938
> @github.com>

Gracias a vos, Javier.